theFork/uMIDI #24: Document apps

theFork, 2015-12-20:


haggl, 2016-03-06:
Github pages ftw:
https://thefork.github.io/uMIDI/

Created with this awesome [tutorial](https://growworkinghard.wordpress.com/2015/11/17/doxygen-documentation-on-github-using-gh-pages/).

haggl, 2016-03-06:
TODO:
- [ ] Complete the application pages  (expression.dox, switcher.dox, ...)
  - [ ] Add some nice pictures
  - [ ] Provide more detailed descriptions
  - [ ] Document GPIO pin mappings
  - [ ] Add links to hardware repos (https://github.com/haggl/uMIDI-hmi, https://github.com/theFork/uMIDI-wah, https://github.com/theFork/uMIDI-switcher)
- [ ] Elaborate hardware section
  - [ ] Maybe move the contents of pin_map.md to Doxygen as well?
  - [ ] Update README (#41)