theFork/uMIDI #48: Add firmware version info to VCP help string

theFork, 2016-03-14:
When entering help into the serial console, it should say something like
`<src/switcher> Build 2015-08-15 git 5b88bde`
I'm afraid that a tough one... But really helpful!

haggl, 2016-03-14:
Hmm I agree, that would be quite helpful. I'll think about it..

haggl, 2016-03-14:
May be not that hard after all:

We could create little script in the toolchain that does the following:
1. Use `git describe --all --dirty --always` to extract SCM information. This generates a string of the form `heads/app/switcher-dirty`.
2. Format the string (remove "heads/", add current date).
3. Generate a C header file containing the resulting string in a `#define`.
This macro could then be used in a `usb_printf()`-statement.

It is also possible to to all this directly in a make recipe.

haggl, 2016-03-14:
19d81ea :+1:

closed 2016-03-14 by haggl