theFork/uMIDI #52: Serial console freezes

theFork, 2016-04-09:
I'm afraid that's a tough one...

In the switcher app, the serial console freezes under the following circumstances:
1. Entering malformed command such as `restorx 01 000...`
2. Using history, i.e. restoring bank 0, then arrow up to restore bank 4 with the same data

- Tested with minicom and the switcher application
- When disconnecting and reconnecting USB communication continues to work. No reset required
- This bug probably affects the library and hence other appliances

haggl, 2016-05-20:
This appears to be a general problem with output of long lines: The console freezes as soon as it prints a line > 77 characters.

The error can be reproduced by simply typing an invalid command longer than 58 characters. The shell then prints "Unknown command: [...]", which in turn freezes the console.
The same holds for long lines in command descriptions. In this case the console freezes when the user wants to see the help message.

haggl, 2016-05-25:
We should just document this properly and close this issue.
There is also a limitation on the overall length of the string, which should also be investigated and documented.

haggl, 2016-06-27:
Got it! :-) It's a stack buffer overflow in `usb_printf()`. *facepalm*

PR #61 (2016-07-23): Fixes for issue52
These changes should fix #52.

There was a pretty stupid buffer overflow in `usb_printf()` and another rogue pointer in the command history code.

/o\

Please test and merge this.
closed 2016-07-23 by theFork