theFork/uMIDI #53: Firmware update via USB

theFork, 2016-04-09:
## Observations
Firmwareupdate via USB does not always work. See the terminal output below:
```
[simon@obelix:switcher]$ sudo make fwupdate 

*** Uploading new firmware...
echo -n "fwupdate `du -b main.bin | sed 's/\s.*//'`\r" > /dev/umidi
sleep 2
cat main.bin > /dev/umidi & cat /dev/umidi
Unrecognized escape sequence!

*z

  E
   Unrecognized escape sequence!

Unknown command: [      
                                                                                                      ��
                                                                ��]
^A

^CMakefile:252: recipe for target 'fwupdate' failed
make: *** [fwupdate] Interrupt

```

- Minicom is running during fwupdate. Here the content with a little context:
```
Unknown command: [~]
Type `help` for help.

fwupdate 19026
Erasing temporary application flash section...
Error erasing temporary application section: 1 
```

## Further information
Properties of the VCP device:
```
[simon@obelix:switcher]$ stty -F /dev/umidi 
speed 115200 baud; line = 0;
min = 1; time = 5;
ignbrk -brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo -echoe -echok -echoctl -echoke
```

OS/Kernel:
```
[simon@obelix:switcher]$ uname -a
Linux obelix 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt20-1+deb8u3 (2016-01-17) x86_64 GNU/Linux
```

theFork, 2016-04-09:
BTW: I'm not using my potentially flash-corrupting USB hub ;)

haggl, 2016-04-09:
> Error erasing temporary application section: 1

According to the [Xboot API documentation](https://github.com/alexforencich/xboot#42-return-values) the error code `1` means that there is something wrong with the Xboot API functions:

> #define XB_ERR_NO_API 1
> [...]
> XB_ERR_NO_API is returned when the loader cannot find the API calls in xboot (either the APIs are disabled, xboot is not installed, or the loader is not looking at the right address)

theFork, 2016-04-17:
Ahhh, ok apparently there was no bootloader installed on the board. Re-Installing it fixed this issue. Closed.

closed 2016-04-17 by theFork