theFork/uMIDI #62: Missing CC values

theFork, 2016-07-23:
Apparently not all possible CC values are transmitted by the expression app. I observed this with echo activated on the serial console, directed the output to a file, sorted the lines and removed duplicate, see the [resulting file](https://github.com/theFork/uMIDI/files/379461/exp.txt).

This is the software installed, it already includes the fixes for #52 :
Software ID: 2016-07-23 expression app/expression-0271e1c

Maybe useful - the current calibration dump:
Offset: 187
Min: 0
Max: 87

haggl, 2016-07-23:
It's not a bug - it's a feature! ^^

The ADC sample input range [0, 87] is scaled to [0, 127] with a simple linear function: 
output = (127 / 87) * input
Since both input and output are integers, some values of the output set are skipped.

What we *could* do is pass the raw ADC samples to the math function without converting them to MIDI values first. See:
file:///home/haggl/repositories/uMIDI/doc/html/adc_8c.html#a544753e4ff62df959e2202025faaa0d9
But then all applications will have to do this themselves at some point.

Another possibility could be using the linear_to_midi function in the above mentioned function directly.

haggl, 2016-07-25:
Fixed: dac0d4b

closed 2016-07-25 by haggl